bigtreetech/BIGTREETECH-Kraken #5: Is there a description of how to use onboard canbus?

altraWeb, 2024-02-03:
There are two canbus ports on this board but i can't find any howto's / Description to use that for example with an EBB SB2209.

Thanks for any help

Jbori347, 2024-02-14:
you need to flash the board with Katapult and set it in usb to can bridge mode.
you can follow instructions here: https://github.com/Esoterical/voron_canbus/tree/0bbde71d092cf38e8d3855190a9cf860a7f238ab

RyanBuzar, 2024-04-28:
> you need to flash the board with Katapult and set it in usb to can bridge mode. you can follow instructions here: https://github.com/Esoterical/voron_canbus/tree/0bbde71d092cf38e8d3855190a9cf860a7f238ab

I flashed catapult per the catapult instructions. I can't get the board out of DFU mode. The DFU mode LED is on, but when I check dfu-utils -l there aren't any controllers in DFU. 

The documentation for this board is awful. I've had to get general information from other user guides to try to piece together information. For a "Premium" board, this is really sub-par documentation.

GKSN, 2024-09-04:
> > you need to flash the board with Katapult and set it in usb to can bridge mode. you can follow instructions here: https://github.com/Esoterical/voron_canbus/tree/0bbde71d092cf38e8d3855190a9cf860a7f238ab
> 
> I flashed catapult per the katapult instructions. I can't get the board out of DFU mode. The DFU mode LED is on, but when I check dfu-utils -l there aren't any controllers in DFU. Only other thing I can do is double-click the reboot button and go into catapult flash mode. Hard reboot or anything doesn't fix the issue.
> 
> The documentation for this board is awful. I've had to get general information from other user guides to try to piece together information. For a "Premium" board, this is really sub-par documentation.

Hi! 
So... I'm having a good learning curve regarding "Kraken and the CAN", until now I wasn't able to make it work but I think I got the direction right this time ( maybe not haha ), the main thing is that I wanted to use the Rpi UART mode comms with the CANBUS function and for what I can see or for what I cannot see it's that it's a little more complicated than I would like... 
Getting back to the problem, I think I had the same problem as you, although I don't know if it is the same problem, the partial solution ( since it is not what will make the CAN BUS work ) is that you have to rewrite the default firmware to the mainboard (available in this repository ). The LED from the DFU Mode is locked in the ON state but I could put the board on DFU Mode as usual, after I followed the instructions from this repository and used the [command](https://github.com/bigtreetech/BIGTREETECH-Kraken/tree/master/Firmware#writing-bootloader-for-kraken-v10): `sudo dfu-util -d ,0483:df11 -R -a 0 -s 0x8000000:leave -D ./Kraken_H723_bootloader.bin` the board came back to life :-). Now I will try to make everything the right way ( or at least try ), for now, I will use the Rpi USB to communicate with the KRAKEN using it as a CAN bridge, by far the best tutorial I saw about is the **[ESOTERICAL GUIDE](https://canbus.esoterical.online/)** mentioned above. 
I hope everything went well with your board, I'm writing all of this since it could help someone like me.
PS: If someone knows how to make the KRAKEN use UART to talk to the Rpi and at the same time work as a CAN bridge please let me know :-).