bigtreetech/BIGTREETECH-Kraken #6: 3D Model Fixing Holes incorrect?

BoothyBoothy, 2024-03-03:
The top two fixing holes appear to be incorrect in the 3D model compared to the Unit I received
[image]
.

Jbori347, 2024-03-27:
Yea I've been asking for an accurate cad for a while now

BoothyBoothy, 2024-03-28:
> Yea I've been asking for an accurate cad for a while now

If of any use, you can find my simplified massing model with the correct holes fixings @ the link below. Best Boothy
https://www.dropbox.com/scl/fi/7l8dt1qe8xlcu022lw5sm/BIGTREETECH-Kraken-V1-Massing-Model-with-correct-hole-fixings-V1.step?rlkey=n1eh09faqckbfyctjiih5ku8a&dl=0