bigtreetech/BIGTREETECH-Kraken #9: Cannot get a BTT Mini 12684 display to work with Marlin on the Kraken

BoothyBoothy, 2024-03-18:
Hi All,
I am testing a Kraken with Marlin on a DIY CNC machine.
I find it useful to have some form of display for ease of tweaking settings when tuning or jogging the axis for general maintenance.
I just cannot get the Mini display or the neo pixels to work at all with the Kraken with Marlin firmware?
I have tried many different combinations in the Marlin firmware including selecting other generic Mini display options with zero response.
On earlier control boards I have used an old BTT touchscreen which has been perfect for my use case but this does not appear possible with no TFT connection for the board?
I am using the latest Kraken Marlin branch form the Marlin git.
Anyone any ideas?
Thanks in advance
Boothy

thisiskeithb, 2024-03-28:
I can confirm that none of the Mini 12864 displays with NeoPixels work on the Kraken under Marlin. The LCD lights up & NeoPixels will cycle on boot, but nothing is displayed. Using [`M250 C255`](https://marlinfw.org/docs/gcode/M250.html) to set max contrast also doesn't work.

Can you [file a bug report in the main Marlin repo](https://github.com/MarlinFirmware/Marlin/issues/new/choose) so this can be tracked? I'll do some more digging to see if I can get it working.

thisiskeithb, 2024-03-28:
Actually, simply add `#define FORCE_SOFT_SPI` anywhere in your config and that will work. I'll submit a PR to fix this bug upstream in Marlin shortly.

h/t to @ellensp for finding that missing define!