bigtreetech/BIGTREETECH-Kraken #11: TMC5160 or TMC2160?

tescolopio, 2024-03-30:
In the documentation you have labeled the stepper drivers as TMC2160 which doesn't match up with the configuration of TMC 5160. This is everywhere in the manual and marlin setup documentation. You can understand my confusion because I cannot find TMC5160 in the klipper documentation but... I can find a stepper drive TMC2160 developed by MKS.

Can you please clarify and correct in either case?

Thanks!

[image]
[image]

tescolopio, 2024-03-30:
Also, you have it listed as 2160 EVERYWHERE. Amazon, your website, advertising. But also it says 5160. Look at this picture. At the top it says 2160. Bottom says 5160. Thoughts?
[image]
[image]

tescolopio, 2024-03-31:
I realize that the architecture of the steppers are indeed TMC2160A-TA. I looked at the schematic. My question, to which I created a ticket with klipper with for as well, is do we need a specific config reference for the TMC2160 or is using the 5160 fine? Does this matter over time? Are there new features offered through the 2160 that should be configurable?

5jvm0u4, 2024-06-27:
I read form somewhere that TMC2160 has the same functionality and pinout as TMC5160, the only difference is TMC5160 has it's own built-in accel-decel control in one specific mode.

5jvm0u4, 2024-06-27:
I read it here, and I'm searching the same question as you are.
https://www.cnblogs.com/xing2/p/17457821.html

looxonline, 2024-06-27:
The drivers are functionally (and internally) identical. The 2160 simply has a part of the silicon that is not used in 3D printing untested and deactivated.

closed 2024-06-27 by looxonline

tescolopio, 2024-06-27:
Thanks for replying all. Much appreciated. I have been running with the 5160 drivers defined in my klipper and have had no issues.