bigtreetech/BIGTREETECH-Kraken #14: Manual needs clarification on Sensorless Homing & endstop pins

HumbleDeer, 2024-05-12:
## Problem

The endstop port's signal pin and the DIAG jumper's signal pin share the same I/O pin on the CPU, through a 100 ohm resistor. 
The manual does not mention this explicitly.

## Demonstration

Both the endstop connectors used for mechanical switches as well as the DIAG pin headers meant for shorting together when using sensorless homing use the same signal line to the processor.

This causes problems when one has shorted this DIAG header for use with sensorless homing, while intending to hook up another device to the endstop "switch" connector. 

In particular, common scenarios are using Normally Open switches going between signal and ground, with a pull-upresistor to VCC. In this scenario, the sensorless homing works fine until the mechanical switch is pressed and the DIAG signal is shorted to ground. Since the mechanical switch may be at the same location as where the sensorless homing triggers, this can cause pretty dramatic behaviour. 

The following excerpts are from the [schematic PDF file](https://github.com/bigtreetech/BIGTREETECH-Kraken/blob/92f19e6f04c5ce2bd3fa09200e5593d87ca9a873/Hardware/BIGTREETECH%20Kraken%20V1.0-SCH.pdf) that can be found under the [Hardware](https://github.com/bigtreetech/BIGTREETECH-Kraken/tree/92f19e6f04c5ce2bd3fa09200e5593d87ca9a873/Hardware) folder:

| Endstop connector | DIAG header |
|-|-|
|[image]|[image]|

# Solution

The most straightforward resolution of this oversight would be to add a disclaimer to the section of the manual that describes the DIAG-pins, noting that the corresponding endstop connector can no longer be used once the DIAG pins have been shorted as instructed and sensorless homing is utilized.

## Notes

This issue was brought to my attention by user @Kobihikri, and documented by myself.

WO1984PR, 2024-07-30:
No response on this I assume. Having a problem with getting sensor less homing to be triggered. Been running in circles.

TrevTrev84, 2024-07-30:
> No response on this I assume. Having a problem with getting sensor less homing to be triggered. Been running in circles.

Set SGT to 1 and speed at 100.... just went through this and guys on Reddit helped!!